fieldOfView/OctoPrint-ngrok #1: Latest Updates Broke Plugin

jneilliii, 2020-06-23:
```
2020-06-23 14:54:06,198 - octoprint.server.api.settings - ERROR - Could not load settings for plugin Ngrok Tunnel (0.1.0)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 288, in _get_plugin_settings
    result = plugin.on_settings_load()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 69, in on_settings_load
    data["auth_pass"] = self._deobfuscate(data["auth_pass"])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 269, in _deobfuscate
    return octoprint.util.to_native_str(zlib.decompress(base64.b64decode(octoprint.util.to_bytes(x))))
  File "/home/pi/oprint/lib/python3.7/base64.py", line 87, in b64decode
    return binascii.a2b_base64(s)
binascii.Error: Incorrect padding
2020-06-23 14:54:06,273 - octoprint.server.api.settings - ERROR - Could not load settings for plugin Ngrok Tunnel (0.1.0)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 288, in _get_plugin_settings
    result = plugin.on_settings_load()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 69, in on_settings_load
    data["auth_pass"] = self._deobfuscate(data["auth_pass"])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 269, in _deobfuscate
    return octoprint.util.to_native_str(zlib.decompress(base64.b64decode(octoprint.util.to_bytes(x))))
  File "/home/pi/oprint/lib/python3.7/base64.py", line 87, in b64decode
    return binascii.a2b_base64(s)
binascii.Error: Incorrect padding
2020-06-23 14:54:06,290 - octoprint.plugins.ngrok - INFO - (Re-)connecting to ngrok tunnel
2020-06-23 14:54:06,291 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin ngrok
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/__init__.py", line 109, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 149, in on_api_command
    self._ngrok_connect()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 208, in _ngrok_connect
    if not (self._settings.get(["auth_name"]) and self._settings.get(["auth_pass"])):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 524, in _func
    return orig_func(*args_mapper(args), **kwargs_mapper(kwargs))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 1519, in get
    result = process()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 1513, in process
    return self._get_value(path, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 1474, in _get_value
    value = preprocessor(value)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 58, in <lambda>
    dict(auth_pass=lambda x: self._deobfuscate(x)),  # getter preprocessors
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 269, in _deobfuscate
    return octoprint.util.to_native_str(zlib.decompress(base64.b64decode(octoprint.util.to_bytes(x))))
  File "/home/pi/oprint/lib/python3.7/base64.py", line 87, in b64decode
    return binascii.a2b_base64(s)
binascii.Error: Incorrect padding
```

jneilliii, 2020-06-23:
[image]

fieldOfView, 2020-06-23:
Since this part is str/bytes sensitive: are you using Python 2 or Python 3?

jneilliii, 2020-06-23:
Python 3.7

fieldOfView, 2020-06-23:
In your config.yaml, is `plugins/ngrok/auth_pass` obfuscated or clear-text? 
Is `plugins/ngrok/_config_version` set?

Do you use any special characters in your password?

fieldOfView, 2020-06-23:
I'm not having luck reproducing this error.

jneilliii, 2020-06-23:
These are the steps that I used to convert my virtualenv.

https://community.octoprint.org/t/upgrade-your-octoprint-install-to-python-3/21232

Settings never save because of the error it seems, nothing in config.yaml, password has no special characters just text and numbers. There was also error on version migration, so I even tried clearing the settings completely. Let me do a full uninstall/reinstall of the plugin and see what happens.

fieldOfView, 2020-06-23:
Ah, I see what is going on: the settings migrations is not working as expected. Fortunately this only affects those who tested my plugin since last friday (so... you?).

fieldOfView, 2020-06-23:
Could you test 90fef2a (the devel branch)?

jneilliii, 2020-06-23:
Yep, that version works.

jneilliii, 2020-06-23:
I just looked at your changes and think it's catching the scenario now where there is a blank auth token, where before sending an empty string to those util functions were breaking.

fieldOfView, 2020-06-23:
Good point, let me test that before I push a release.

jneilliii, 2020-06-23:
That was my first test, by clearing the data during uninstall. Be careful though because there is a bug with OctoPrint that if you clear the data it can leave a `ngrok: null` in config.yaml which causes all kind of binding headaches.

fieldOfView, 2020-06-23:
Done some more testing with partial configurations. Tagging 0.1.1

closed 2020-06-23 by fieldOfView

foosel, 2020-06-23:
> Be careful though because there is a bug with OctoPrint that if you clear the data it can leave a `ngrok: null` in config.yaml which causes all kind of binding headaches.

For the record, that bug will be fixed with 1.4.1.

fieldOfView, 2020-06-23:
Thanks again!

(I swear I don't always fumble around like this :pleading_face:)

commit 90fef2a1 (2020-06-23): Fix error loading setting in migration step

Fixes #1