fieldOfView/OctoPrint-ngrok #4: allow use of custom domains

Indochild56, 2020-06-25:
It would be awesome to use a completely custom domain, or at least the option to.

fieldOfView, 2020-06-26:
I will certainly add this option in a future version; it is something that ngrok supports, but *only for paid accounts* (from $5 per month). Two caveats:

* I would like to keep the configuration form compact, without too many options a user may not (be able to) use. Some rearranging may be in order. 
* I'll have to look in to how this works with SSL certificates; will they register a certificate for you so you automatically have working HTTPS?

fieldOfView, 2020-06-26:
Are you aware that custom domains are only available on business and pro accounts? They also require a bit of external configuration on your domains, which this plugin cannot do for you, and the ssl certificate will be broken.

Indochild56, 2020-06-26:
I have done the first two, just looking into the ssl certificate part.

your two caveats makes sense though.

fieldOfView, 2020-06-27:
I have implemented the functionality here: https://github.com/fieldOfView/OctoPrint-ngrok/commit/e1146a856dd8ce7bcf1b7244eb7f98fd50d0eb33. But since I don't have a pro account I can't test it.

Indochild56, 2020-06-27:
I'll test it out as soon as its released.

fieldOfView, 2020-07-13:
0.2.0 is now available

closed 2020-07-13 by fieldOfView