fieldOfView/OctoPrint-ngrok #5: AstroPrint comparability

ongo-gablogian, 2020-06-25:
It seems, at least on my instance that astroprint and ngrok tunnel cannot be active simultaneously, ngrok tunnel takes over and AstroPrint account connection is removed. Trying to restore astroprint account link results in an error until ngrok tunnel is disabled.

fieldOfView, 2020-06-26:
I don't know exactly how AstroPrint works and I don't have it installed, but I can't think of how the presence of a tunnel could affect AstroPrint. The ngrok tunnel cannot "take over"; a tunnel does not actively block or redirect traffic through other channels, it just offers a new channel to reach OctoPrint from outside. Traffic from OctoPrint to eg the AstroPrint servers should just take the original route and be blissfully unaware of the tunnel. 

It might help if you say what the error you encounter says (exactly). Your OctoPrint logs may include other hints (for me).

fieldOfView, 2020-07-13:
I can't do much unless you respond to my remarks above, specifically:

> It might help if you say what the error you encounter says (exactly). Your OctoPrint logs may include other hints (for me).

fieldOfView, 2020-07-27:
Closing due to inactivity.

closed 2020-07-27 by fieldOfView