fieldOfView/OctoPrint-ngrok #6: Custom SubDomain not working

Indochild56, 2020-06-26:
Custom subdomain not working on reserved subdomain, It continues to put in a random subdomain even though i'm on a paid account and have the subdomain reserved.

Not sure if i have to put in the subdomain with ".ngrok.io"?

fieldOfView, 2020-06-26:
I admit I have not tested the option, since I don't have a paid account myself. I'm just using the option available via their API.

I'll shell out for a paid account and investigate. Thanks for reporting!

fieldOfView, 2020-06-26:
It turns out this was a typo, which is easily fixed here: b639fb7

This fix will be part of the next release.

fieldOfView, 2020-06-27:
I like to keep the issue open until there is a released version that includes the fix.

fieldOfView, 2020-07-09:
@Indochild56 and @inomad360, could you help me test this development version?
https://github.com/fieldOfView/OctoPrint-ngrok/archive/devel.zip

You can install it via OctoPrint Settings -> Plugin Manager -> Get more -> ... from URL. It should fix this issue, but it also implements more granular permissions to start/stop the tunnel. Would love to know if that works for you.

inomad360, 2020-07-09:
I will install today and let you know. Thank you.

inomad360, 2020-07-09:
It works....you are amazing. thank you

Indochild56, 2020-07-12:
Works Great! sorry for the delay!

inomad360, 2020-07-12:
One more question/ enhancement idea. Is it possible to create a tunnel when
octopi boots? I remotely turn the printer on/off. When octopi comes online,
the default seems to be no tunnel. Since I have a paid domain, I have o be
in the network o create it. Thank you.

fieldOfView, 2020-07-12:
@inomad360, I have opened a new issue for you: #12. Please don't ask questions that are unrelated to the current bug, or I will loose track and important reports will go missing.

fieldOfView, 2020-07-13:
0.2.0 is now available

commit b639fb70 (2020-06-26): Fix setting subdomain option

Fixes #6
closed 2020-07-13 by fieldOfView