fieldOfView/OctoPrint-ngrok #7: Easy access to random tunnels

theely, 2020-07-02:
Thanks a lot for the plug-in, awesome idea!
Have you thought about an easy way to retrieve the randomly generated tunnel address on external devices?
For example would it be possible to push the tunnel address somewhere? Could be as simple as sending an email.

fieldOfView, 2020-07-02:
Thought about it, yes. Decided on how to do it: not yet ;-)

The tunnel address is already available through https://dashboard.ngrok.com/status/tunnels. The easier I make the access to the tunnel address, the less secure everything gets. Also, once I have implemented one way (eg email, as you suggest), people will want their preferred way added (eg push to telegram, some SMS service, etc). That would become a maintenance nightmare, and introduce many opportunities to leak the tunnel address.

If you get a paid ngrok account (I have no affiliation with them, I swear!), you can get a fixed tunnel address.

theely, 2020-07-24:
Alternatively you could trigger an event when a new tunnel is established.
There are several plugins that allow to perform action on events:
- https://plugins.octoprint.org/plugins/IFTTT/
- https://plugins.octoprint.org/plugins/growl/
- https://plugins.octoprint.org/plugins/slack/

fieldOfView, 2020-07-24:
Yes, I could fire a custom event when a tunnel is established. However, out of the plugins you listed, I think only IFTTT supports custom events directly. And it is still a bit akin to installing a fancy lock on your frontdoor and telling all your housemates - whom you know can't keep a secret - how to duplicate the key. But I'll add a custom event in a future version; good suggestion.

fieldOfView, 2020-07-27:
The master branch now fires the `plugin_ngrok_connected` event when the plugin creates a tunnel, and the `plugin_ngrok_closed` event when a tunnel is closed.

fieldOfView, 2020-07-27:
Please help testing the fix by installing the current development version via OctoPrint Settings -> Plugin Manager -> Get more -> ... from URL with the following url: https://github.com/fieldOfView/OctoPrint-ngrok/archive/devel.zip

commit 63bfff75 (2020-07-27): Fire custom events plugin_ngrok_connected and plugin_ngrok_closed

Fixes #7
closed 2020-07-30 by fieldOfView