fieldOfView/OctoPrint-ngrok #8: Camera

kawaczek, 2020-07-04:
I have two cameras in ocroprint. How to set them to be displayed in the tunnel?

fieldOfView, 2020-07-04:
Currently the best (only?) way to do this with the plugin is if you use a reverse proxy (such as haproxy) to make OctoPrint and both the cameras available through the same http port. This is how OctoPi comes configured by default (though only for a single webcam).

A future version of the plugin will let you register additional tunnels (up to a total of 4 with a free ngrok account). including tunnels to your additional camera(s) and possibly additional OctoPrint instances if you have multiple.

jneilliii, 2020-07-31:
@fieldOfView this looks like the current webcam issue thread, I just noticed with the NGROK tunnel the default steam isn't loading. I'm using an OctoPi image with default haproxy config and when accessing the site via the NGROK tunnel, the URL `/webcam/?action=stream` is not seen. I get the webcam stream not loaded error and clicking the link in that message it appears to be going to the correct address with a 404 not found. Any suggestions?

fieldOfView, 2020-09-07:
@jneilliii, this thread is more about hosting multiple tunnels. I'll deal with your issue here: #18 (now that I understand what is going wrong).

Ingegneus, 2023-04-28:
I found that this still doesn't work with octo4a. is this only me or is there a config step I have overseen? the tunnel works fine for the main octoprint app, but I can't see the embedded stream, from outside my local network

TheRealFalseReality, 2025-04-20:
If you have more than 1 webcam, does any webcam come through at all? I expected the default one, after changing port from 5000 to 80 in ngrok, still, no camera.