fieldOfView/OctoPrint-ngrok #9: Use with Printoid app on android

mpoffo1231, 2020-07-07:
Sorry if this is in the wrong place.  I cannot seem to get this configured to work with Printoid on my Android phone.  I am sure there is something I must be doing incorrectly but I cannot seem to find any guide or instructions anywhere to assist. Can you assist or point me in the right direction?

mpoffo1231, 2020-07-07:
Figured it out.    Put your Ngrok link/address in the Public IP or Hostname area and then enable Basic Authentication in the advanced settings and put in your Ngrok login credentials.

fieldOfView, 2020-07-07:
Good to know you got it going!

mpoffo1231, 2020-07-07:
Well I may have spoken a bit too soon.  I have two printers that I want to use.  I can only have 1 simultaneous connection to Ngrok.  The links seem to change even when octoprint does not reboot.  So to get a solid connection it looks like I need to pay the 60/year?  Is there a way to set this up with 2 printers on a free account?  Will a paid account allow to not have 2 change the settings constantly?

fieldOfView, 2020-07-08:
> I have two printers that I want to use. I can only have 1 simultaneous connection to Ngrok.

This is a current limitation of the plugin. A future version will have the ability for one OctoPrint instance to setup and "share" up to 4 tunnels on a free ngrok account. There are a couple of things you can do now:
* adjust the haproxy configuration to forward requests to the second instance (eg octopi/webcam2 and octoprint/instance2)  
* use multiple ngrok accounts, one for each OctoPrint instance
* (try) to use different regions for each instance

The latter may be the easiest to try, and I am not sure if it works but during testing it looked like you could have a tunnel per region open.

With a free account, the tunnel address will change each time you save the settings of the plugin (because that recreates the tunnel). A paid account will let you reserve a subdomain to use, so there will not be any changes. 

Unfortunately, even with the 60/year account you will not currently be able to use 2 tunnels on two different raspberry pis. Setting up a reverse proxy on one of the instances is your best bet until I create the version that shares tunnels from a single raspberry pi.

mpoffo1231, 2020-07-13:
Can you provide some direction on how to forward.  Are you talking about doing port forwarding in my router using the unique link/web address ngrok provides?

I tried to multiple accounts method but it did not seem to work.  I could get only one working at a time.  It appears the service recognizes that I have the same IP as the other account and simply will not work.  Perhaps I did something wrong but thought I had it configured correctly.

How do I set up different regions?

Looking forward to your updated plugin that allow you to share tunnels.  :)

Learning this stuff still, so your help is appreciated.

fieldOfView, 2020-07-14:
> How do I set up different regions?

The tunnel settings in OctoPrint has a `region` dropdown where you ca choose between United States, Europe, Asia/Pacific, etc. I can run two instances, one in the United States and one in Europe, from the same ngrok account.

fieldOfView, 2020-07-14:
I am closing this issue, because you answered your own question of how to use it with Printoid. Feel free to continue to discuss here though.

fieldOfView, 2020-07-14:
The issue of hosting multiple tunnels will be tracked in #8

closed 2020-07-14 by fieldOfView