fieldOfView/OctoPrint-ngrok #10: Using a paid Ngrok account with Reserved subdomain with the plugin

inomad360, 2020-07-09:
Amazing plug-in and thank you. I wanted a fixed ngrok subdomain to access my octopi and therefore got a paid basic ngrok account. I have my sub domain defined in the ngrok dashboard. It is also entered into the plug-in settings as a sub domain. However the plug-in continues to assign a random subdomain as if I do not have a paid account. 

How do I have the plug-in connect with my sub domain that is defined in the settings? Thank you

[image]

fieldOfView, 2020-07-09:
I'm marking this as a duplicate of https://github.com/fieldOfView/OctoPrint-ngrok/issues/6. I will shortly update that issue with a development snapshot for you to test/use. Sorry for the inconvenience.

closed 2020-07-09 by fieldOfView

fieldOfView, 2020-07-09:
Please see this comment: https://github.com/fieldOfView/OctoPrint-ngrok/issues/6#issuecomment-656079814