fieldOfView/OctoPrint-ngrok #11: Use self hosted ngrok

cconde, 2020-07-11:
Hi,

I have a self hosted ngrok server I would like to use:
https://github.com/inconshreveable/ngrok/

This requires to add my server and port and maybe the option to trust self signed certificates as stated at:
https://medium.com/@shivanshvij/a-self-hosted-ngrok-45bcec06d0cf

I can test it upon demand.

Thanks and great job!

fieldOfView, 2020-07-12:
The server you found is an old version of ngrok ("v1"), which is not compatible with the current version that is used by ngrok.com ("v2"). The current version in use at ngrok.com has not been released as open source. From the project readme: 

> Both the client and server are known to have serious reliability issues including memory and file descriptor leaks as well as crashes.

I would love to do this - because I don't like to rely on a centralized 3rd party service either - but I don't know if ngrok v1 is worth the significant development effort to support.

fieldOfView, 2020-07-12:
The goal of this plugin is to have something that works out of the box; it should be more secure than port forwarding, while being easier to set up. Self-hosted ngrok v1 fails at both. For now I'm going to close this issue because of the limited "return" on the development effort required to support the ngrok v1 client.

I am following other tunnel providers with interest. https://lynk.sh/ is currently in beta, but promises a free self-hosted version once it gets out of beta.

closed 2020-07-12 by fieldOfView