fieldOfView/OctoPrint-ngrok #12: Creating tunnels on start

fieldOfView, 2020-07-12:
Is it possible to create a tunnel when
octopi boots? I remotely turn the printer on/off. When octopi comes online,
the default seems to be no tunnel. Since I have a paid domain, I have o be
in the network o create it. Thank you.

fieldOfView, 2020-07-12:
@inomad360, do you have the `Create a tunnel when OctoPrint starts` checkbox checked? That should create a tunnel when OctoPrint starts. If it doesn't, that is a bug that needs to be investigated.

inomad360, 2020-07-12:
I will test tomorrow and let you know.

inomad360, 2020-07-13:
I have tested and it does NOT automatically create tunnel on startup. I have unplugged the server 2x and restarted. I can confirm that I have checked the "Create a tunnel when OctoPrint starts checkbox" checked? I am on 0.2.0 and using a custom domain. The custom domain fix works great.

fieldOfView, 2020-07-13:
Then I'll have to investigate why this is not working. Could you upload your octoprint.log for me? Just in case something unexpected is visible there.

inomad360, 2020-07-13:
here it is

fieldOfView, 2020-07-13:
Unfortunately, attachments don't carry over via github. So you would either have to drag/drop it into a comment here in the github interface, or post a link to somewhere on dropbox, google drive or the likes.

inomad360, 2020-07-13:
[octoprint log.log](https://github.com/fieldOfView/OctoPrint-ngrok/files/4914216/octoprint.log.log)

fieldOfView, 2020-07-14:
This is a strange error in your log:
```2020-07-13 17:00:02,021 - pyngrok.process - ERROR - t=2020-07-13T17:00:02+0100 lvl=eror msg="failed to reconnect session" obj=csess id=996fbe31bf2a err="dial tcp: lookup tunnel.us.ngrok.com on 192.168.86.1:53: dial udp 192.168.86.1:53: connect: network is unreachable"```

It looks like ngrok can't access the ngrok server because the network is unreachable when OctoPrint starts.

I am grasping at straws here, but I see that you also have the Astroprint plugin installed, and there is at least one user that says the two plugins are somehow incompatible: https://github.com/fieldOfView/OctoPrint-ngrok/issues/5 (but I don't know how/why). Could you try disabling that plugin to see if that makes any difference?

inomad360, 2020-07-14:
Thanks for checking the log. Uninstalled Astroprint and rebooted. Ngrok did not create a tunnel upon startup and error is there in the log . Here is what I tried:
1. ping  192.168.86.1  - ping is OK
2. ping tunnel.us.ngrok.com - timeout
3. ping 3.20.27.198 - timeout
4. ping 192.168.86.1:53 - timeout

Will further diagnose today.

[octoprint.log](https://github.com/fieldOfView/OctoPrint-ngrok/files/4918510/octoprint.log)

inomad360, 2020-07-14:
One more data point. Ngrok is the only plugin that cannot access the internet.

[image]

fieldOfView, 2020-07-14:
There are strange things going on in your network. From the latest log you posted:
```
2020-07-13 11:44:54,504 - octoprint.plugins.pluginmanager - ERROR - Could not fetch plugins from repository at https://plugins.octoprint.org/plugins.json: HTTPSConnectionPool(host='plugins.octoprint.org', port=443): Max retries exceeded with url: /plugins.json (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb0183e10>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',))
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 988, in _fetch_repository_from_url
    r = requests.get(repository_url, timeout=30)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/raven/breadcrumbs.py", line 341, in send
    resp = real_send(self, request, *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
ConnectionError: HTTPSConnectionPool(host='plugins.octoprint.org', port=443): Max retries exceeded with url: /plugins.json (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb0183e10>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',))
```

```
2020-07-13 11:44:55,112 - octoprint.plugins.softwareupdate - WARNING - Could not check fullscreen for updates due to a network error
2020-07-13 11:44:55,121 - octoprint.plugins.softwareupdate - WARNING - Could not check astroprint for updates due to a network error
2020-07-13 11:44:55,124 - octoprint.plugins.softwareupdate - WARNING - Could not check simpleemergencystop for updates due to a network error
2020-07-13 11:44:55,125 - octoprint.plugins.softwareupdate - WARNING - Could not check PrusaETAOverride for updates due to a network error
2020-07-13 11:44:55,129 - octoprint.plugins.softwareupdate - WARNING - Could not check octoprint for updates due to a network error
2020-07-13 11:44:55,138 - octoprint.plugins.softwareupdate - WARNING - Could not check octolapse for updates due to a network error
2020-07-13 11:44:55,141 - octoprint.plugins.softwareupdate - WARNING - Could not check ngrok for updates due to a network error
2020-07-13 11:44:55,141 - octoprint.plugins.softwareupdate - WARNING - Could not check TheSpaghettiDetective for updates due to a network error
```
(so not just ngrok)

fieldOfView, 2020-07-14:
> 
    1. ping 192.168.86.1 - ping is OK
    2. ping tunnel.us.ngrok.com - timeout
    3. ping 3.20.27.198 - timeout
    4. ping 192.168.86.1:53 - timeout

1 seems to be your router, which can be reached. That's good.
4: you can't ping a specific port, so this is expected not to work.
2 and 3: Well, if ngrok can't reach its servers, it cannot work. You could try the other regions (Europe for example: `tunnel.eu.ngrok.com` or `52.28.187.147`).

I am again grasping a straw: could it be that it takes a while after booting before your Pi has a reliable connection to your wifi?

inomad360, 2020-07-14:
I uninstalled and installed ngrok plugin, still the same error. The plug in works, but it does not connect to the custom subdomain initially on startup even when the button is checked. I am not a coder here  -  but how does the plugin create a tunnel after startup if it does not have access to ngrok servers?

[octoprint.log](https://github.com/fieldOfView/OctoPrint-ngrok/files/4918936/octoprint.log)

fieldOfView, 2020-07-14:
> how does the plugin create a tunnel after startup if it does not have access to ngrok servers?

I have no idea.

Could you try pinging `tunnel.eu.ngrok.com` and `52.28.187.147`?

fieldOfView, 2020-07-16:
This is not an isolated issue (though I still think the problem is somehow in your network/router):
https://community.octoprint.org/t/ngrok-tunnel-a-more-secure-but-simple-to-use-alternative-to-port-forwarding/21224/46?u=fieldofview

It looks like I may have to add a "retry" mechanism if the first connection to the ngrok service fails.

inomad360, 2020-07-16:
To diagnose the issue, I have re-flashed and re-installed octoprint. Still will not login upon restart. Log attached. Thank you.

[octoprint.log](https://github.com/fieldOfView/OctoPrint-ngrok/files/4931731/octoprint.log)

inomad360, 2020-07-16:
...also, since it connects without any issues after the reboot is complete, can there be a check to see if it is logged in with a time lag?

fieldOfView, 2020-07-27:
Commit 37405be adds a retry after 20 seconds. This is in the `devel` branch.

fieldOfView, 2020-07-27:
Please help testing the fix by installing the current development version via OctoPrint Settings -> Plugin Manager -> Get more -> ... from URL with the following url: https://github.com/fieldOfView/OctoPrint-ngrok/archive/devel.zip

inomad360, 2020-07-27:
Seems to work at boot. Tried twice. Thank you so much.

fieldOfView, 2020-07-27:
Great, thanks for testing. Of course the greater issue remains to fix your network while starting, but that is beyond the scope of this plugin. Good to know it works for now.

closed 2020-07-30 by fieldOfView

inomad360, 2020-07-30:
Thank you. Works very well.