fieldOfView/OctoPrint-ngrok #13: Webcam access

dzid26, 2020-07-16:
I am currently accessing webcam stream at http://octopi.lan:8080/?action=stream and octoprint at http://octopi.lan:5000/.
The webcam stream doesn't work using ngrok. 

What is the requirement to access the webcam? - Does the port need to be changed? Or is haproxy needed?

Thank you

fieldOfView, 2020-07-16:
Currently, only a single tunnel is supported to haproxy is needed so you can access both OctoPrint and the webcam on the same port.

A future version of the plugin will let you set up multiple tunnels (up to 4 with a free ngrok account).

fieldOfView, 2020-07-16:
Functionally, this is a duplicate of #8

closed 2020-07-16 by fieldOfView

kadrik0, 2023-08-11:
I ran into a similar issue. I am using the octopi image from raspberry pi imager (the new forked one with better webcam support). My classic stream stream url for the camera is /webcam/?action=stream. I started off with port 5000 for ngrok based on a youtube video I found but I kept getting a 404 error body= message when using octapp on ios. When I changed the local port in octopi’s ngrok tunnel page from 5000 to 80 it fixed my issues. Posting here in the hopes it helps someone else too.

TheRealFalseReality, 2025-04-20:
Damn, not working for me...