fieldOfView/OctoPrint-ngrok #14: Optionally run a script after tunnel opened?

derekantrican, 2020-08-25:
I have a ngrok free plan which means that every time a tunnel is created the url is randomized. Instead of paying for a ngrok plan, I'd like to write a script to do something with assigned ngrok tunnel url (eg email it to myself, use an API to set it as a url redirect with my personal domain, etc).

Simply - could there be an option added to launch a script (with the newly generated ngrok tunnel url as an argument) once the ngrok tunnel has been created?

fieldOfView, 2020-08-25:
See https://github.com/fieldOfView/OctoPrint-ngrok/issues/7. The plugin fires the `plugin_ngrok_connected` event when the plugin creates a tunnel, and the `plugin_ngrok_closed` event when a tunnel is closed. Both events have the tunnel address as the payload. You could use IFTTT to send emails, or another plugin that responds to plugin events..

derekantrican, 2020-08-25:
Ok, I will look into the IFTTT plugin. Thanks!

derekantrican, 2020-08-25:
Ok, I've got the IFTTT plugin installed and successfully triggering off the `plugin_ngrok_connected` event. What should I put in for "Value1", etc to also pass the tunnel url as a parameter?

[image]

derekantrican, 2020-08-25:
Looking at the [Value syntax](https://github.com/tjjfvi/OctoPrint-IFTTT/wiki/Value-syntax) for the IFTTT plugin it seems that it is expecting JSON/dict for the event payload but the ngrok plugin is only sending the single value of the tunnel url. I can start a PR suggesting this change

derekantrican, 2020-08-27:
Closing with the merge of #16

fieldOfView, 2020-09-15:
I reopened the issue, so I can keep track of changes. I'll soon release version 0.3.0 (in testing now in the `devel` branch)

closed 2020-09-17 by fieldOfView