fieldOfView/OctoPrint-ngrok #15: Address in "Tunnel status" does not include "https://" prefix

derekantrican, 2020-08-25:
The address listed in the plugin settings under "Tunnel status" does not include the "https://" prefix. This is a problem because if you simply copy the contents of that box and paste it into a browser you will get the error: `Tunnel [subDomain].ngrok.io not found`. If the "https://" was included this would not be an issue

fieldOfView, 2020-08-25:
Right next to the address, to the left of the "Close" button, there is a link button. You can click it to visit the tunnel link, or right-click and select Copy link location to get the full tunnel address.

derekantrican, 2020-08-25:
Good to know - I didn't know you could right click that button. I think my suggestion still stands

closed 2020-09-17 by fieldOfView