fieldOfView/OctoPrint-ngrok #18: Make port configurable

fieldOfView, 2020-09-07:
@jneilliii reported that the camera is not working for him. _Originally posted by @jneilliii in https://github.com/fieldOfView/OctoPrint-ngrok/issues/8#issuecomment-667309001_

> @fieldOfView this looks like the current webcam issue thread, I just noticed with the NGROK tunnel the default steam isn't loading. I'm using an OctoPi image with default haproxy config and when accessing the site via the NGROK tunnel, the URL `/webcam/?action=stream` is not seen. I get the webcam stream not loaded error and clicking the link in that message it appears to be going to the correct address with a 404 not found. Any suggestions?

The ngrok plugin looks at what `publicPort` is configured to use in the `discovery` section of `config.yaml`. For OctoPi, this is normally set to port 80:
https://github.com/guysoft/OctoPi/blob/58ad94db1db2c8eb104f07c5cfab50c57da5916b/src/modules/octopi/filesystem/home/pi/.octoprint/config.yaml#L9
For manual installs, it is normally set to port 5000. Port 5000 skips the reverse proxy, so the webcam "redirect" is also requested from port 5000. 

This should be solved by making the port configurable in the plugin (with a default to what `publicPort` is configured to)

closed 2020-09-17 by fieldOfView

Ingegneus, 2023-04-29:
I found that using a single camera still doesn't work with octo4a. is this only me or is there a config step I have overseen? the tunnel works fine for the main octoprint app, but I can't see the embedded stream, from outside my local network

fieldOfView, 2023-04-29:
The plugin works on one configurable port. If you want to see a webcam (or multiple thereof), they will have to be served from the same port. You would need a reverse proxy for that, such as HAProxy that gets installed on OctoPi (the raspberry pi distribution) by default. I don't know if Octo4a includes any reverse proxy.

Ingegneus, 2023-04-29:
puh ok that is all way over my head 😅