fieldOfView/OctoPrint-ngrok #22: Error "your account is limited to 1 simultaneous ngrok client sessions" on create tunnel

derekantrican, 2020-09-23:
There seems to be some situations where restarting octoprint (or maybe other situations) does not kill/close the current ngrok session. If there is an error when trying to create a tunnel similar to "your account is limited to 1 simultaneous ngrok client sessions" then the plugin should kill the running ngrok process then try again.

fieldOfView, 2020-10-30:
Unfortunately, this is hard to do. The plugin only has access to the tunnel that the plugin itself created. If a plugin is somehow "orphaned", it is no longer under the control of the plugin, so the plugin can not close it. I will look in to guiding the user to ways to close the tunnel via https://dashboard.ngrok.com

derekantrican, 2020-11-10:
Can you run terminal commands from the python extension? If so, I would just run `killall ngrok` which would kill any currently running ngrok process

pacineee, 2022-12-16:
This ngrok issue is getting me a headache... a moment I'm getting a solution...

aplahore4, 2023-02-17:
In windows I killed the process from task manager and run `ngrok http 8000` It worker for me.

dannyboothman, 2024-02-17:
1. Login to [ngrok dashboard](https://dashboard.ngrok.com/)
2. Go to: Tunnels > [Agents](https://dashboard.ngrok.com/tunnels/agents)
3. You'll then see a list of active agents, and on the right of each agent, 3 icons: **Stop**, **Restart**, and **Update**. Click **Stop**.

blarzHernandez, 2024-06-05:
> `ngrok http 3000 --region us`

This allowed me having more than 1 simultaneous client sessions

codingram2003, 2024-07-05:
i didn't find any active agents running in the ngrok dashboard. Still it is showing me the error