fieldOfView/OctoPrint-ngrok #23: Showing [tunnel not connected] even though tunnel is created successfully on startup.

Hooch180, 2020-11-07:
Hi.

My Toolbar shows Circle with cross and message [tunnel not connected] even though in ngrok panel I see that tunnel got created just after powering raspberry pi. And I can connect via it too.

Here are parts of log with ngrok errors.
[ngrok_log_parts.txt](https://github.com/fieldOfView/OctoPrint-ngrok/files/5505274/ngrok_log_parts.txt)

fieldOfView, 2020-11-08:
It looks like this may be caused by a breaking change in the python bindings for ngrok. I'll look into a fix tomorrow.

Hooch180, 2020-11-08:
@fieldOfView - Thank you very much. Looking forward for update.

fieldOfView, 2020-11-09:
Should be fixed with 708f287

I'll keep this issue open until I publish a fixed version.

closed 2020-11-09 by fieldOfView