fieldOfView/OctoPrint-ngrok #24: When loading plugin, "ngrok binary was not found. Be sure to call "ngrok.install_ngrok()" first for "ngrok_path""

earthicko, 2020-11-16:
There seems to be a problem when loading the plugin initially. Here's the error log below:

> 2020-11-16 19:19:01,783 - octoprint.plugin.core - ERROR - There was an error loading plugin ngrok
> Traceback (most recent call last):
>   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1114, in load_plugin
>     plugin.load()
>   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 378, in __plugin_load__
>     __plugin_implementation__ = NgrokPlugin()
>   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok/__init__.py", line 37, in __init__
>     ngrok.get_version()
>   File "/home/pi/oprint/lib/python3.7/site-packages/pyngrok/ngrok.py", line 359, in get_version
>     ngrok_version = process.capture_run_process(pyngrok_config.ngrok_path, ["--version"]).split("version ")[1]
>   File "/home/pi/oprint/lib/python3.7/site-packages/pyngrok/process.py", line 357, in capture_run_process
>     _validate_path(ngrok_path)
>   File "/home/pi/oprint/lib/python3.7/site-packages/pyngrok/process.py", line 376, in _validate_path
>     "\"ngrok_path\": {}".format(ngrok_path))
> pyngrok.exception.PyngrokNgrokError: ngrok binary was not found. Be sure to call "ngrok.install_ngrok()" first for "ngrok_path": /home/pi/oprint/lib/python3.7/site-packages/pyngrok/bin/ngrok

Adding `ngrok.install_ngrok()` on line 35 of "./__init__.py" solved the problem, though I'm not sure this is a valid solution.

fieldOfView, 2020-11-16:
`ngrok.install_ngrok()` should already be called when installing the plugin. Could you check/share your plugin_pluginmanager_console.log?

earthicko, 2020-11-16:
Here it is:
> 2020-11-16 18:54:34,673   /home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpkapd4lg_/OctoPrint-ngrok-master.zip --no-cache-dir
> 2020-11-16 18:54:37,096 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> 2020-11-16 18:54:37,097 > Processing /tmp/tmpkapd4lg_/OctoPrint-ngrok-master.zip
> 2020-11-16 18:54:37,097 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-ngrok==0.3.1) (1.4.2)
> 2020-11-16 18:54:40,855 > Collecting pyngrok>=4.1.3
> 2020-11-16 18:54:40,855 > Downloading https://www.piwheels.org/simple/pyngrok/pyngrok-5.0.0-py3-none-any.whl (21 kB)
> 2020-11-16 18:54:40,856 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (0.5.0)
> 2020-11-16 18:54:40,856 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (1.1.0)
> 2020-11-16 18:54:40,856 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (1.1.2)
> 2020-11-16 18:54:40,856 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (1.2.0)
> 2020-11-16 18:54:40,856 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (1.0.1)
> 2020-11-16 18:54:40,856 > Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (0.1.5.post0)
> 2020-11-16 18:54:40,856 > Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (5.7.2)
> 2020-11-16 18:54:40,856 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (0.18.2)
> 2020-11-16 18:54:40,857 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (0.8.0)
> 2020-11-16 18:54:40,857 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (2.11.2)
> 2020-11-16 18:54:40,857 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (3.1.1)
> 2020-11-16 18:54:40,857 > Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (2020.8.7)
> 2020-11-16 18:54:40,857 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (2.0)
> 2020-11-16 18:54:40,857 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (0.6.0)
> 2020-11-16 18:54:40,857 > Requirement already satisfied: feedparser<6,>=5.2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (5.2.1)
> 2020-11-16 18:54:40,857 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (1.1.1)
> 2020-11-16 18:54:40,858 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (2020.9.27)
> 2020-11-16 18:54:40,858 > Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (1.12.1)
> 2020-11-16 18:54:40,858 > Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (3.4)
> 2020-11-16 18:54:40,858 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (5.3.1)
> 2020-11-16 18:54:40,858 > Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (2.24.0)
> 2020-11-16 18:54:40,858 > Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (1.2)
> 2020-11-16 18:54:40,858 > Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (0.4.21)
> 2020-11-16 18:54:40,858 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (1.0.7)
> 2020-11-16 18:54:40,859 > Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (0.1.1)
> 2020-11-16 18:54:40,859 > Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (0.57.0)
> 2020-11-16 18:54:40,859 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-ngrok==0.3.1) (1.5.0.1)
[... cut ...]
> 2020-11-16 18:54:42,198 > Stored in directory: /tmp/pip-ephem-wheel-cache-ngd3a42p/wheels/a3/c4/a1/6390dc6a995e165c0d276071a2b7fc3780e741f2364ae9587c
> 2020-11-16 18:54:42,198 > Successfully built OctoPrint-ngrok
> 2020-11-16 18:54:42,198 > Installing collected packages: pyngrok, OctoPrint-ngrok
> 2020-11-16 18:54:42,198 > Successfully installed OctoPrint-ngrok-0.3.1 pyngrok-5.0.0

It's rather strange cause I can't find any error or warning logs. I've set octoprint's logging level to 'DEBUG'.

fieldOfView, 2020-11-16:
It turns out the ngrok binary should automatically be downloaded and installed the first time `ngrok.connect()` is called, so any errors would not show up in the plugin installer log. So the bug at hand is that `ngrok.connect()` is not the first call to the binary that happens (`ngrok.getVersion()` is).

closed 2020-11-16 by fieldOfView

fieldOfView, 2020-11-16:
Fixed in version 0.3.2. Thanks for the report!

earthicko, 2020-11-16:
Thank you for your fast support! I appreciate your activeness and commitment.

commit d20d8119 (2020-11-16): Fix checking pyngrok version

Fixes #24