fieldOfView/OctoPrint-ngrok #25: Plug-in now showing in Octoprint

soundforascene, 2020-11-16:
Since updating to 0.3.2 my it does not seem to be functioning. I have gone through the logs and and found this. 

|  OctoPrint-ngrok (0.3.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_ngrok
2020-11-16 14:12:24,490 - octoprint.plugins.pluginmanager - INFO - The plugin was installed successfully: OctoPrint-ngrok, version 0.3.2
2020-11-16 14:12:24,795 - octoprint.plugins.tracking - INFO - Sent tracking event install_plugin, payload: {'plugin_version': '0.3.2', 'plugin': 'ngrok'}
2020-11-16 14:12:34,615 - octoprint.plugin.core - ERROR - Invalid syntax in /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_ngrok/__init__.py for plugin ngrok
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_ngrok/__init__.py", line 36
    if ngrok.__version__ >= '5.0.0'

I am not entirely sure if this is due to the update or my own wrong doing. Hope it helps.

jneilliii, 2020-11-16:
@fieldOfView it looks like you're missing a colon in your if statement...

https://github.com/fieldOfView/OctoPrint-ngrok/blob/65c58e0839c0e9cdf1a83ad72a7c4a42756af47e/octoprint_ngrok/__init__.py#L36

fieldOfView, 2020-11-16:
> it looks like you're missing a colon in your if statement...

It certainly looks that way... 🤦‍♂️

fieldOfView, 2020-11-16:
Fixed in 0.3.3

closed 2020-11-16 by fieldOfView