fieldOfView/OctoPrint-ngrok #35: Update Agent

trulow, 2023-12-19:
Ngrok is going to discontinue access for any application using ngrok agents version 3.1 and older on January 15th, 2024

Will this plugin be updated to support the newest agent?

arontaupe, 2024-01-31:
solved - the plugin still works with the updated agent. simply ssh into your host and update ngrok :)

MarcinKnD, 2024-02-16:
how to update a ngrok thats its implemented in plugin? it is not regullary insalled but hided in bin venv.

its not respond for update comand, max I have is 2.3.41. Its not responding for ngrok config upgrade. 

some one can help?

jneilliii, 2024-02-18:
another possible option is running this command
```
/home/pi/oprint/bin/ngrok update
```

fieldOfView, 2024-02-18:
Hey, it's been a while since I touched this plugin (and OctoPrint plugins at all). Since you've given me a good reason to push an updated version of the plugin (#37), this issue would also be solved by upping the pyngrok version in setup.py (https://github.com/fieldOfView/OctoPrint-ngrok/blob/master/setup.py#L36) for a new version of the plugin, right?

MarcinKnD, 2024-02-18:
@fieldOfView It will be great, because grok plugin is grate for those who don't wanna pay for remote control of orctoprint.

I update the pyngrok like @jneilliii said it is updated to 5.1 version and stil nothing. Still Octoprint warning that version is 2.3.41 and can't be used. I also try the second command but it didn't work (I try one more time later this day)

jneilliii, 2024-02-18:
> 5.1 version and stil nothing

this is strange because the most recent versions are 

ngrok version 3.6.0
pyngrok version 7.1.2

got this by running 

```
/home/pi/oprint/bin/ngrok -v
```

so I think either you're on a python version that doesn't support the latest version of the python module or you need to run the update command
```
/home/pi/oprint/bin/ngrok update
```

I did this last night on my pi zero 2w and it took a long time to complete, but once done it upgraded from 3.5.0 agent to 3.6.0 agent.

jneilliii, 2024-02-18:
> issue would also be solved by upping the pyngrok version in setup.py (https://github.com/fieldOfView/OctoPrint-ngrok/blob/master/setup.py#L36) for a new version of the plugin, right?

not sure how that works with pyngrok, and would have thought the pip install --upgrade command would have straightened it out, but seems it didn't. until we know more about @MarcinKnD's environment might not make sense to add that pinning.

jneilliii, 2024-02-18:
oh one other question @MarcinKnD did you restart OctoPrint after running the `/home/pi/oprint/bin/pip install --upgrade pyngrok` command?

jneilliii, 2024-02-18:
according to another user on the forum the other command worked for them

https://community.octoprint.org/t/ngrok-agent-too-old-ngrok-error-121/57001/4?u=jneilliii

jneilliii, 2024-02-18:
maybe pinning to version 6 or higher would do it based on the release notes...still doesn't explain why the pip upgrade didn't get there. 

https://github.com/alexdlaird/pyngrok/releases/tag/6.0.0

jneilliii, 2024-02-18:
...and based on these release notes, you wouldn't want to go higher than 7.0 for octopi 0.18 support. 

https://github.com/alexdlaird/pyngrok/releases/tag/7.1.0

jneilliii, 2024-02-18:
So it seems that there is no support here for octopi 0.18. I got on my old instance and tried all the different update methods mentioned before and nothing seems to work to get the version of the agent above 2.3.41. Probably best option at this point is to create a backup of OctoPrint, download backup, flash latest stable OctoPi version and restore the backup during initial setup wizard. Installing the plugin on a fresh OctoPi image didn't seem to have any issues getting the newer agent when I installed.

jneilliii, 2024-02-18:
ok, so opened a ticket on the module repo and got a very quick response on how to update. If you run these commands in SSH session it should work for you. I think I may have had the older version on that 0.18 instance from before. 

```
source /home/pi/oprint/bin/activate
rm /home/pi/oprint/lib/python3.7/site-packages/pyngrok/bin/ngrok
ngrok -v
deactivate
```

this is because the ngrok update command only works for minor version updates.

progebytes, 2024-02-25:
> ok, so opened a ticket on the module repo and got a very quick response on how to update. If you run these commands in SSH session it should work for you. I think I may have had the older version on [...]

@jneilliii thank you so much, i finally found your thread! these commands are the solution at the problem! Thank you so much for your work :)

MarcinKnD, 2024-02-27:
> oh one other question @MarcinKnD did you restart OctoPrint after running the `/home/pi/oprint/bin/pip install --upgrade pyngrok` command?

ok. sorry for delay. I used  `/home/pi/oprint/bin/pip install --upgrade pyngrok` followed by @progebytes comands and now I have 

```
ngrok version 3.6.0
pyngrok version 7.1.3
```

and now it working 

Thanks guys